BjarneBitscrambler/OrientationSensorFusion-ESP #1: Support

DanielG86, 2020-10-27:
Hello Bjarne,
Great project and just wanted to let you know that I’m willing to support where possible. 
Currently just prepared a test version on a breadboard, with some issues, as reported in the PR (I think #166)

BjarneBitscrambler, 2020-10-29:
Hi Daniel, 
Good to have another person interested in this sensor! I read your comments over on the SensESP repository, and I think that clarifies the drifting heading issue a bit. Like @tedenda and you, I had seen the reading wander off intermittently, but didn't know why. Your observation that it was related to USB wire proximity, and that recalibration corrected it, suggests that the fusion algorithm is having trouble coping with distortions in the magnetic environment.

There's a couple things you could try that would clarify/confirm this (perhaps you have already tried them):
- does moving the wire back to its original position restore correct operation? Or does it always require recalibration once the drifting starts?
- does the interference happen with wires that aren't carrying any current?
- what is the approximate safe distance between the sensor and the wire?
- if the wire is fixed in location nearby, and calibration is performed, then does the heading continue to stay stable until the wire is moved? (I believe this is what you've already observed)

If you want, I can give you edit permissions on this repository so you can enter your comments/notes in the Wiki section. Let me know if that would be useful - otherwise I'll just copy your comments from this issue into the Wiki.

DanielG86, 2020-10-29:
Hello Bjarne,
I was quite tied up in another problem with my SignalK set-up, although I think I've pinpointed the source of my problems; unrelated to this sensor.
When I have some more time, I'll look into your questions, and do some testing. Edit permissions is not needed, I prefer to respond this way

BjarneBitscrambler, 2020-10-29:
Further on this issue, my reading of NXP's change log for the fusion algorithm suggests that the latest version might be more robust that the one we've currently tried. It's version 4 that the Adafruit AHRS library pulls in, but that's from approx 2015. I've downloaded version 7.20 from NXP, and a few of the improvements that might be useful for us are:
- magnetic calibration can be performed by the fusion algorithm itself (i.e. running a separate routine with an external tool is not needed)
- calibration of the accelerometer and gyro is supported
- calibration can be updated simultaneously with running of the fusion algorithm (I believe what this means is that if the magnetic environment of the sensor changes, the algorithm can dynamically adapt to it. Maybe.)
- the fusion software streams packets of sensor and performance data to a serial port, and this can be read by an NXP Windows program that they say is helpful when developing sensors applications - it plots readings, displays orientation as determined by different algorithm variations, etc, etc.  I haven't tried the tool yet, and I don't know if it's limited to just Windows, but it does sound useful when exploring sensor performance.

I've been working through adapting the NXP fusion source to compile for the ESP platforms (NXP has released it under BSD-clause 3 license, so it's free for porting).  Currently I have it compiling successfully and am working on replacing the I2C functions that are written for NXP processors with the equivalent calls on the ESP.  Might take another two weeks, but then we should be able to compare the existing software with the newer fusion routines.

I haven't bothered posting any of the work-in-progress fusion software, but if you are interested let me know and I can push it up to this repository.

-

DanielG86, 2020-10-29:
Sounds great improvements, but content-wise a little over my head. I’ll stick to testing, when you have something available 😎

DanielG86, 2020-10-30:
> Hi Daniel,
> Good to have another person interested in this sensor! I read your comments over on the SensESP repository, and I think that clarifies the drifting heading issue a bit. Like @tedenda and [...]

I did some testing this morning. First impressions: when I move the USB line (from the ESP) close to the sensor, it does interfere. I've observed a small deviation (10 degrees or so), coming back to original values after removal, but then a constant number, even when the sensor was turned. In other words: it doesn't give correct measures anymore, it froze.. I've also seen a sawtooth behavior after bringing the USB line close. A reset of the ESP solves the problem, and brings the sensor back to normal behavior. Another try showed shifting heading with a close USB line, but steady and correct numbers after removing the USB line. 'Close' is like 1-2 centimeter, so fairly close. I guess if you enclose the sensor in a fairly large enclosure, you shouldn't suffer from this anymore. All tests were done on my desk, with all kinds of other electronics, but no observable interference. Sensor only moved around the heading axis, so no pitch and roll.
A non-current bearing USB line did nothing to the readings. I guess the metal volume is way too small to have influence. 

All together a pretty good behavior of the sensor, as long as you can avoid close proximity of electric currents. The USB current is pretty small, so a large current (e.g. bow truster) would have impact from a larger distance. But this is common for magnetometers

BjarneBitscrambler, 2020-11-01:
Thanks Daniel for that info - I have added it to the Wiki.

closed 2020-11-01 by BjarneBitscrambler