BjarneBitscrambler/OrientationSensorFusion-ESP #5: Confirm that the Sensor Axes are oriented correctly

BjarneBitscrambler, 2020-11-04:
Testing with the **Sensor Toolbox** indicates that the axes of the sensors might not be oriented correctly. It appears that the X and Y axes of both the magnetometer and the accelerometer are swapped. The file `hal_axis_remap.c` needs to be edited to correct this. The expected orientation for nautical use (and apparently aeronautical) is called *NED*, where X,Y,Z are in a right-hand coordinate system pointing *North, East, and Down* respectively.

BjarneBitscrambler, 2020-11-05:
Corrected the X,Y swaps of the axes in `hal_axis_remap.c` 

There is still an issue wherein the Sensor Toolbox shows the sensor rotating the opposite direction around the Z-axis than it really is.  As well, the angular readout on the compass heading is inverted (e.g. 90 degrees shows up as 270 degrees).

Current functions in `hal_axis_remap.c` do not apply any operations to the axes.

BjarneBitscrambler, 2020-11-06:
Completed adjusting and testing the axes - the current software works with the Adafruit AXOS8700/AXAS21002 combination sensor breakout board.  The following comments are in `hal_axis_remap.c` and repeated below for convenience.

With an unknown sensor board, the most straightforward way to determine the correct axis transformations is with the help of the NXP Sensor Toolbox. 
- In the remapping functions set the axis transformations to your best guess (or start with no transformations at all). 
- Load the software, launch the Toolbox, and then	observe the orientation readouts while manipulating the board.
- It helps to go through one sensor mode at a time, starting with the magnetometer (since it is an absolute reference, unlike the other two)
-- Using the Digital Compass algorithm (which doesn't rely on accel or gyro), remap the axes as needed to get the Heading displaying correctly (i.e. near 0 degrees when your board is flat and pointing north, and the heading increases with clockwise rotation). 
-- Next adjust the accelerometer axes, using the Pitch and Roll readouts together with the Tilt-Compensated Compass algorithm. For nautical use, convention has roll increasing to starboard, and pitch increasing with the bow pointing up.
-- Finally, adjust the gyro	axes by ensuring that rotations in all three axes give the expected response (i.e. that the magnetometer and accel are not 'fighting' the gyro, which causes the display to jerk around).
The most common transformations are inverting (changing the sign) and swapping axes (e.g. X <-> Y). Theoretically, scaling could be done here too, but there are other places in the software where that makes more sense.

@tedenda and @ DanielG86 : have you had a chance to try out the latest software?  With the WiFi streaming, it's easier to test things without the USB cable getting wrapped....  Results this afternoon seemed good - I was getting quick response and stability on the heading reading after ensuring the magnetometer was calibrated (the Toolbox tab that shows magnetometer performance and cal readings is very useful, and the calibration can be uploaded to the ESP with a button-click.)

closed 2020-11-06 by BjarneBitscrambler

DanielG86, 2020-11-10:
Haven't had the time to look at your progress in the past days. And still learning how to use Github, so basically I don't know (yet) how to use your new code, unless it's merged in the SensESP main branch. Will dive into that first, I guess.

DanielG86, 2020-11-10:
mmm, you wrote a really good readme, which explains how to upload the code! Will take a chance whenever I have the time

BjarneBitscrambler, 2020-11-11:
Great!  Don't hesitate to ask me any questions that come up.  Consider
yourself a guinea pig :-)

I haven't worked on the new fusion code for a couple days, but it should be
in a functional state. My next plan is some housekeeping: renaming files,
functions, etc...

Cheers,
Bjarne